mikemountain/nfl-led-scoreboard #2: KeyError 'prehome'

sklafg, 2020-08-27:
Hi,

I getting an error when i try to execute the program.

sudo python main.py --led-gpio-mapping=adafruit-hat
INFO (17:03:07): NFL Scoreboard - v0.0.0a (64x32)
INFO (17:03:09): Scheduled State, waiting 12h
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 33, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 40, in __render_game
    self._draw_pregame()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 123, in _draw_pregame
    home_team_logo_pos = self.screen_config.team_logos_pos[overview['hometeam']]['prehome']
KeyError: 'prehome'

mikemountain, 2020-08-27:
ah, looks like I missed out on committing some code when I made this. I'll try to find some time soon to push the necessary code, will definitely have it done before the season starts

sklafg, 2020-08-27:
That will be great. Thanks!

mikemountain, 2020-09-09:
Think it should be fixed now. Give'r a pull and let me know if you still get a problem!

sklafg, 2020-09-09:
It is fixed. :-) Thanks!
I am getting an error with the logos now. 

No such file or directory: logos/Ari.png.

Maybe the problem is the .PNG in uppercase.

sklafg, 2020-09-09:
I renamed the logos. It works fine now.

mikemountain, 2020-09-10:
Ah shit, thanks! I'm doing a real poor job of committing these changes when I fix them. Gotta stop working on this late at night. thanks!

closed 2020-09-10 by mikemountain