mikemountain/nfl-led-scoreboard #3: Broken touchdown animation

haynesa, 2020-09-11:
Super cool project, looking forward to having it up all season. Had it up for last nights game, but the touchdown animation appears to be broken. Any thoughts?

mikemountain, 2020-09-11:
How do you mean broken? Is there a stacktrace you can post or anything?

plarsen007, 2020-09-13:
I think I know what he is talking about - 

I would get the following after every score; 
INFO (13:21:24): Refresh game overview
INFO (13:21:24): TD
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 29, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 55, in __render_game
    self._draw_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 164, in _draw_game
    self._draw_td()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 285, in _draw_td
    ball = Image.open("assets/td_ball.gif")
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2312, in open
    fp = builtins.open(filename, "rb")
IOError: [Errno 2] No such file or directory: 'assets/td_ball.gif

I fixed it by just renaming Assets directory to lower case a; assets. 

that resolved the break for me.

mikemountain, 2020-09-13:
gah, thank you @plarsen007. Looks like git didn't rename the folder when I pushed that last, doesn't seem to like case changes. Will try to tweak that and push it up asap

mikemountain, 2020-09-13:
Closed after pushing up the filename change. If things continue to break post-pull, re-open

closed 2020-09-13 by mikemountain