mikemountain/nfl-led-scoreboard #4: No module named tzlocal

Bombsurace, 2020-09-12:
**Describe the bug**
I am trying to run the start up command and getting the ImportError: No module named tzlocal

I have double checked my timezone is set up on my end locally and all that.  I reinstalled everything and still get the same error.

**To Reproduce**
Steps to reproduce the behavior:
I use start up code sudo python main.py --led-gpio-mapping=aduafruit-hat-pwm --led-brightness=75 --led-slowdown-gpio=2

Even just the basic sudo python main.py, but non the less it just says No module named tzlocal.  

Did I miss something in the install or code that I need to deliberately go in and change myself?

mikemountain, 2020-09-12:
Weird. Can you run `pip list` and tell me if tzlocal shows up? If it doesn't, can you tell me if the `install.sh` script has the line `sudo pip install pytz tzlocal requests`? If it doesn't, just run that command and try to re-run

Bombsurace, 2020-09-12:
Looked through the install.sh and it did not have sudo pup install pytz tzlocal requests in it.  So I ran that then re-ran the main.py, this time with an ImportError: No module named lxml

mikemountain, 2020-09-12:
Can you post the full stack trace? And can you run a `git pull` as well just in case

Bombsurace, 2020-09-12:
Wont lie, not sure how to do that? New to a lot of this, can get into files and search around code, but still learning the bigger stuff.

When I type git pull it posts error: cannot open .git/FETCH_HEAD: Permission denied

Bombsurace, 2020-09-12:
strace I am guessing what I need to use? what - should I use?

mikemountain, 2020-09-12:
No problem man, and no need for strace, just copy and paste the command line output from the spot where you run the command to the end of the error.

The git issue might be if you have initially run it with sudo, so if you run `sudo git pull ` I'm sure it'd work

Bombsurace, 2020-09-12:
Already up to date.
 so that should be good

Bombsurace, 2020-09-12:
Traceback (most recent call last):
  File "main.py", line 5, in <module>
    from data.data import Data
  File "/home/pi/fantasy-football-scoreboard/data/data.py", line 3, in <module>
    import sleeper_api_parser as sleeper
  File "/home/pi/fantasy-football-scoreboard/data/sleeper_api_parser.py", line 4, in <module>
    from lxml import html
ImportError: No module named lxml

mikemountain, 2020-09-12:
Well there's one issue solved - this is the NFL scoreboard repo, not the fantasy football repo! To fix this you'll need to run `sudo apt-get install python-lxml` and it should work. I think I need to fix a script over there as well if this is an issue.

Bombsurace, 2020-09-12:
I thought that could be the case, but I was following the instructions from the nfl scoreboard page so figured maybe it just ran off the same.install lol
May need to update the nfl scoreboard main page as install instructions have the fantasy football install.  Appreciate the help, will run after golf tourney

mikemountain, 2020-09-12:
Oh damn, you're right, thanks. I tried to build this project even more quickly than the fantasy football one and essentially just hot-swapped pieces out to get it working during the playoffs last season. Thank you for pointing this out!

Bombsurace, 2020-09-13:
Hey no worries man just trying to go through it glad I could help.  I definitely got whoevers fantasy week 1 up and going on the screen but clearly isn't the nfl, so the codes worked, but again its fantasy football and not NFL lol. 

I am sure you are busy and at least the updated lines of install got the LED working and hopefully the NFL side will be up to entere correctly as well.   No rush good sir, I appreciate your work and love what you're doing.  Im sure im just an idiot and am not sure what code to substitute out for NFL scoreboard vs fantasy football scoreboard.  Although I will try exactly that and see what happens this evening

Bombsurace, 2020-09-13:
Done and done,  nfl-led-scoreboard..  im a dumb. all good my friend thanks again for the help! excited to see how this works out

closed 2020-09-13 by Bombsurace