mikemountain/nfl-led-scoreboard #5: Make the logos look nicer

mikemountain, 2020-09-13:
**Is your feature request related to a problem? Please describe.**
like 95% of the logos look like shit

**Describe the solution you'd like**
make them look less shit

**Describe alternatives you've considered**
there's no alternative

**Additional context**
I know how to do it, I just haven't gotten around to it yet because it requires photoshop, but I'll get around to it by week 2. just putting this in to track things

Bombsurace, 2020-09-13:
I have photoshop. Let me know what you need and I can clean up some PNG files

Bombsurace, 2020-09-14:
Yea I could see that having some issues at times,  Any kind of background what so ever that may have a pixel or the like would pop up on the Matrix itself.

Again if you need help, let me know what size you need and I can bust some out and send your way with the transparent backgrounds and all

mikemountain, 2020-09-15:
thanks for the offer @Bombsurace but I took care of it, was pretty quick! appreciate it though

closed 2020-09-15 by mikemountain