mikemountain/nfl-led-scoreboard #6: Enhancement Request: Rotate through all games

ergoalkeeper, 2020-09-14:
**Is your feature request related to a problem? Please describe.**
No, the project currently works as expected (thanks!). Just wanted to put in an enhancement request. 

**Describe the solution you'd like**
A rotation through that day's scheduled NFL games, much like the NHL scoreboard. I'm assuming branching out to show the Thurs and Mon scores might be a bit difficult, but would be great plus to have. Not sure if the API score pull source automatically changes over when the new week begins. 

**Describe alternatives you've considered**
Configuration wise, I'm thinking this would need to be a flag or something set in the config.json file to enable rotation. Maybe like 'fav team = ALL'  meaning rotate through displaying all teams? Or two additional entries for 'rotation = true' and 'time = 15' which enables the rotation code and allows a customization how long each game is displayed before transitioning to the next game to display (like the MLB scoreboard). Not sure if the NHL rotation code is easily accessible or how simple it would be to change/add this feature.  

**Additional context**
Love what has been done so far and I know this will be a work in progress all season. Just wanted to put in a enhancement request which I hope is being considered to be added down the road. Thanks doing this. Made watching the Pack stomp my Vikes a little better with this NFL scoreboard display.

mikemountain, 2020-09-14:
Thanks dude! This is actually on my to-do list, can't promise it'll be done super quickly but I'm hoping to get this set up soon, just have to juggle between this repo and my fantasy football repo, but I should be able to make something off the NHL/MLB repos pretty quickly

siebe41, 2020-09-17:
I 2nd this request. Its a nice feature the NHL one has (They added a lot of features this year).

machinasdei, 2021-01-03:
3rd this feature request!  would be awesome, appreciate what this is now and what its becoming!

mikemountain, 2021-09-02:
only took me a year but I finally (attempted) to implement it with this branch https://github.com/mikemountain/nfl-led-scoreboard/tree/f/rotate-games

am going to wait to see how it fares during week 1 before I merge it into master (gotta clean up all the comments too, blah blah blah) but for anyone wanting to rotate games, they should check that branch!

mikemountain, 2021-09-02:
Nah man, don't worry about it. Running `git fetch`, `git pull` and then `git checkout f/rotate-games` should give you minimal headaches. You might need to run `git stash` after fetching if you've made changes that git complains about, which it probably will re: the config file (which is my fault).

There'll be config changes to make, you can just edit the config.json.example and then save it as config.json. The earlier one won't work anymore

If you want to go back to the main branch, you just need to run `git checkout master`

pjockey, 2021-09-02:
Mike, trying to git fetch, git pull and then git checkout f/rotate-games, get a permission error.  running from nfl-led-scoreboard.

mikemountain, 2021-09-02:
@pjockey can you post the output?

arjunlala, 2021-09-03:
@pjockey First check if you have any local changes with `git status` Chances are if you ran off master it'll show a bunch of changed files. Either way all good.
What you'll have to do is run a `git stash` first to put away your local changes. And then once you do that you can run `git checkout -r ` to see all the available branches (not a necessary step)
From there you should be able to run `git checkout f/rotate-games` and be all set and run the program. Same process for switching to any other branch.
Explanation may have been unnecessary but wanted to give the full picture. Hope it helps (:

arjunlala, 2021-09-03:
@mikemountain Just wanted to say thank you for this amazing build. Currently putting it together for the upcoming season. One question for the `while_preferred_team_live` flag: what exactly do the `enabled` and `during_halftime` flags do? I could wait til we're in game but figured I'd ask. Thanks!

mikemountain, 2021-09-03:
@arjunlala thank you! I hope it works well, but it's been a bit of a mess so we'll see how it goes 😅

Those options I need to keep working at in the code, not properly implemented yet, but they're supposed to control if the board will rotate to other teams while your preferred team is live.

If you turn on rotate teams, and you have NO as your favourite team for example, it will only rotate if they're not playing, and will stay on their game if they are (or, with the second option, if their game is at halftime)

pjockey, 2021-09-03:
> @pjockey can you post the output?
```
git fetch
error: cannot open .git/FETCH_HEAD: Permission denied
```

mikemountain, 2021-09-03:
run `sudo chown -R $(whoami) .git/`

Seems like git ended up owned by root, so this will just set the permissions back to your user and you'll be able to do normal git operations

ergoalkeeper, 2021-09-07:
Loaded the rotate-games branch last night, looks good! Looking forward to give it the initial run this weekend. The retro/8-bit helmets as logos is a cool update as well. Cheers!

WashCaps33, 2021-09-07:
If I start over from scratch will all of these new additions be included?  I am not saavy enough to understand branches and git and such.

ergoalkeeper, 2021-09-07:
@WashCaps33 Honestly, that's what I did. It only took a couple minutes and the json config file isn't that long (easy to reconfigure my fav team quick, along with the new config fields). But you have to add in the step that @arjunlala referenced `git checkout f/rotate-games` to get the right branch/version. 
1) Remove the entire NFL scoreboard on the Rasp Pi (assuming you know where on the Pi the folder is located)
```
sudo rm -Rf nfl-led-scoreboard
```
2) Run the install steps (per norm, takes a couple minutes)
```
git clone --recursive https://github.com/mikemountain/nfl-led-scoreboard
cd nfl-led-scoreboard/
sudo chmod +x install.sh
sudo ./install.sh
```
3) Switch it to the rotate-games branch
```
git status
git stash
git pull
git fetch
git checkout f/rotate-games
```
4) Configure the .json file wihtih the `nfl-led-scoreboard` folder on your Rasp Pi (should have rotate fields now)
```
sudo nano config.json
```

5) Run per norm (with whatever flags you need)
```
sudo python main.py 
```

6) Enjoy!
[image]

mikemountain, 2021-09-07:
well there's no need to blow away everything! it'd be much easier to do these steps:
1. in the repo directory, type `git stash`
2. then type `git fetch`
3. then type `git pull`
4. then type `git checkout f/rotate-games`
5. then type `nano config.json` and edit the fields as you want them
6. run the project as it ran before

WashCaps33, 2021-09-07:
I think I have done it.  I run it and it doesn’t crash, however, all it does is this.  Nothing projects to the LED board

mikemountain, 2021-09-08:
@WashCaps33 can you post the exact output from the console when it runs? I can't really troubleshoot without knowing the output

WashCaps33, 2021-09-08:
Sorry…by output do you mean what appears on the board?  I’m not good at this stuff if you couldn’t already tell.

mikemountain, 2021-09-08:
No problem dude, I'll make sure we get it all figured out. I just want to see what prints out in the console when you run 
`sudo python main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2`

For example, the default behaviour should be this:
```INFO (15:01:37): NFL Scoreboard - v0.0.1 (64x32)
INFO (15:01:39): Pre-Game State
INFO (15:01:41): ping render_game```

Is this exactly what prints out on your console when you run it? Or does it print anything different?

WashCaps33, 2021-09-08:
That’s exactly what comes out when I run the command you supplied.

ergoalkeeper, 2021-09-10:
Seems like the rotate-games branch broke sometime during the game tonight, getting this output now postgame:

[image]

mikemountain, 2021-09-15:
Script should now rotate through all games (and logos should all be where they belong)

closed 2021-09-15 by mikemountain