mikemountain/nfl-led-scoreboard #7: Pre game logos centering

Bombsurace, 2020-09-15:
[image]

Is there a way to get the logos in the pre game more centered on each side? Seems to be off to the right on both sides. 

In game they are perfectly aligned and all that.

mikemountain, 2020-09-15:
Can you pull master just to make sure you've got the latest stuff? I've been tweaking the logo placement config because the logos end up funky like that, the TEN Pre-away x value should be 1.

Bombsurace, 2020-09-15:
Looks good, spacing doesn't look to overlap anymore

mikemountain, 2020-09-15:
Great, thanks. I'm gonna keep this issue open because I'm positive other teams will have messy pre-game placements so this can be for people to notify me if they catch something funky

Bombsurace, 2020-09-16:
[image]

Just pulled it up this morning seeing when the update for the new week starts . Jag logo looks like the "bad" logos from before still and with the blurry on the edges dips onto the VS part

mikemountain, 2020-09-20:
hey @Bombsurace, do you still have this issue? The logo currently looks okay on my board, do you have the latest version pulled?

Bombsurace, 2020-09-22:
I'll try and check sometime tonight or tomorrow after work. It still looks like the "bad" logo from before. 

I did restart my scoreboard this morning to double check, is that the same as a new "pull" or do I need to run and update on my side?

mikemountain, 2020-09-22:
Unless you have a start-up script that does a pull then no, you'll have to run `git pull` to get the latest code

Bombsurace, 2020-09-22:
Got it, may have to add a git pull into my start up script then.  

Logo is updated and looks solid (at least on the post game screen currently).  Thank you .  I'll keep an eye out for any other logos that maybe be wonky .  Thank you good sir

Bombsurace, 2020-09-23:
[image]

Jags home logo pre game a little off? Like you said before maybe just 1 over? Looks clean though. Not sure why it says Sundays at 10 for a Thursday game but the countdown normally starts on time.

mikemountain, 2020-09-23:
yeah, just need to play with the pre-home and pre-away values for both teams in the 64_32config file. I definitely messed up the gametime calculation somewhere along the way because there are a ton of issues with it, I'll try to take a look at this tonight and figure it out

mikemountain, 2021-09-15:
Logos have been standardized across all teams, and should be centred properly now as of the latest change

closed 2021-09-15 by mikemountain