mikemountain/nfl-led-scoreboard #10: Gametime referenced before assignment

habeckmatt, 2020-09-18:
I'm getting the following error:

Traceback (most recent call last): File "main.py", line 30, in <module>                                                                                           MainRenderer(matrix, data).render() 
File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 29, in render self.__render_game()                                                                                                       File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 36, in __render_game self._draw_pregame()                                                                                                       File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 110, in _draw_pregame date_text = gametime.strftime('%A').upper()                                                                              UnboundLocalError: local variable 'gametime' referenced before assignment  

I just cloned the repo within the last day and received this error. Any suggestions on what I can do?

siebe41, 2020-09-18:
I got this last night.

Quick fix is go into the renderer folder, open the main.py file and on line 110 change the variable from gametime to gamedatetime. 

I made a pull request for it.

https://github.com/mikemountain/nfl-led-scoreboard/pull/8

mikemountain, 2020-09-18:
Thanks @siebe41, @habeckmatt I just merged in his PR so it should work now

closed 2020-09-18 by habeckmatt