mikemountain/nfl-led-scoreboard #11: ValueError: Extra data: line 2 column 15

habeckmatt, 2020-09-18:
File "main.py", line 25, in <module> config = ScoreboardConfig("config", args)                                                                                  File "/home/pi/nfl-led-scoreboard/data/scoreboard_config.py", line 8, in __init__ 
    json = self.__get_config(filename_base) 
File "/home/pi/nfl-led-scoreboard/data/scoreboard_config.py", line 29, in __get_config
  reference_config = self.read_json(filename)
File "/home/pi/nfl-led-scoreboard/data/scoreboard_config.py", line 22, in read_json
   j = json.load(open(path))
File "/usr/lib/python2.7/json/__init__.py", line 291, in load
  **kw)
File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
  return _default_decoder.decode(s) 
File "/usr/lib/python2.7/json/decoder.py", line 367, in decode
  raise ValueError(errmsg("Extra data", s, end, len(s)))
ValueError: Extra data: line 2 column 15 - line 5 column 1 (char 15 - 46)

I just got this error after the update.

mikemountain, 2020-09-18:
Can you post your `config.json` contents?

habeckmatt, 2020-09-18:
I accidentally deleted a bracket. Closing the issue!

closed 2020-09-18 by habeckmatt