mikemountain/nfl-led-scoreboard #13: No module named 'nfl_api_parser'

habeckmatt, 2020-11-12:
Traceback (most recent call last):
  File "main.py", line 6, in <module>
    from data.data import Data
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 3, in <module>
    import nfl_api_parser as nflparser
ModuleNotFoundError: No module named 'nfl_api_parser'

This is the error message being returned. Any clue what this is?

mikemountain, 2021-01-03:
hey man, sorry for the mega delay on getting to this - do you still have this issue? I'm wondering if the packages didn't get installed properly

breid1313, 2021-01-30:
@habeckmatt any chance you were running in python3 by mistake? this project is written in python2, the running in python3 will give you these types of import exceptions

mikemountain, 2021-09-15:
Should be fixed with the latest master changes

closed 2021-09-15 by mikemountain