mikemountain/nfl-led-scoreboard #14: Incorrect team abbreviation for Washington

steve-cicco, 2020-12-08:
Noticed during the Steeler game today (Dec 7) that it wasn't working.  Did some digging and found you have Washington's logo and formatting under WAS and it needs to be WSH according to ESPN's API data.

Great job on this.  Added some code on my own to display the red zone animation at Heinz Field:

https://twitter.com/scicco21/status/1331799516857708544?s=20

Despite the horrible loss today...go Steelers!

mikemountain, 2020-12-08:
Ah thanks man, this season's been a bit (just a bit) of a shitshow, as has 2020, so I haven't had as much time to dedicate to working on this project as I would have liked. Really hoping to tighten it up more soon

steve-cicco, 2020-12-09:
It has indeed.  Which is why I'm all the more appreciative that folks like you take time to create stuff like this to keep our minds off everything else.

It also doesn't help that there are only a few hours each week where you can develop with live gameday data.

machinasdei, 2021-01-03:
Hello, I'm having issues selecting Washington as my favorite team and getting the main.py to run.
I've tried WAS and WSH and even WFT in the config.json and neither abbreviation will launch the board.  What should be used?

some quick output fails;
Traceback;
#with WAS;
    gametime = datetime.strptime(self.data.game['date'], "%Y-%m-%dT%H:%MZ")
TypeError: 'NoneType' object has no attribute '__getitem__'

#with WSH
KeyError: u'WSH'

#with WFT;
    gametime = datetime.strptime(self.data.game['date'], "%Y-%m-%dT%H:%MZ")
TypeError: 'NoneType' object has no attribute '__getitem__'

#with WASH;

INFO (11:01:26): NFL Scoreboard - v0.0.1 (64x32)
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/me/nfl-led-scoreboard/renderer/main.py", line 29, in render
    self.__render_game()
  File "/home/me/nfl-led-scoreboard/renderer/main.py", line 33, in __render_game
    gametime = datetime.strptime(self.data.game['date'], "%Y-%m-%dT%H:%MZ")
TypeError: 'NoneType' object has no attribute '__getitem__'

steve-cicco, 2021-01-03:
It should be WSH in the config but you also have to change WAS.png to WSH.png in the logos directory if memory serves

machinasdei, 2021-01-03:
Thanks Steve,
I did catch the .png renames and changed as suggested, but still generates an error;
INFO (13:37:08): NFL Scoreboard - v0.0.1 (64x32)
INFO (13:37:12): Scheduled State, waiting 30
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/scott/nfl-led-scoreboard/renderer/main.py", line 29, in render
    self.__render_game()
  File "/home/scott/nfl-led-scoreboard/renderer/main.py", line 40, in __render_game
    self._draw_pregame()
  File "/home/scott/nfl-led-scoreboard/renderer/main.py", line 116, in _draw_pregame
    awaysize = self.screen_config.team_logos_pos[overview['awayteam']]['size']
KeyError: u'WSH'

mikemountain, 2021-01-03:
In `renderer/64x32_config.json`, can you try changing WAS to WSH there? I'll try to find some time soon to fix this on the main branch

machinasdei, 2021-01-03:
hi mike, thanks, that did the trick.  I missed it in renderer/64x32_config.json
it's working now, got the WSH vs PHI screen saying Today 8:20 PM
thanks for your work on this board!

PR #15 (2021-01-03): fix washington abbr change
fixes #14
closed 2021-01-03 by mikemountain