mikemountain/nfl-led-scoreboard #17: displaying wrong favorite team for playoffs

machinasdei, 2021-01-08:
got a weird one here, so it displays the game Saturday for 1:05PM for IND vs BUF , but my favorite team is set to WSH - which did work vs the iggles last week.  Why would it display the wrong teams for the preview of the next weeks' game?

mikemountain, 2021-01-08:
ah sorry about that, I started this last year during the playoffs so I just wanted to keep testing it with whichever team was currently playing/playing next - also, I think it might crash if a fav team is set that's not in the playoffs. gimme a few mins to make a quick patch branch

mikemountain, 2021-01-08:
hey @machinasdei, let me know if this fixes the issue - if not feel free to reopen this issue and berate me!

closed 2021-01-08 by mikemountain

machinasdei, 2021-01-08:
Hey man, i'll try out the b/fix playoff teams now!  Just wanted to note that I tried changing the favorite team to anyone in the current playoffs and it kept showing that IND vs BUF game

machinasdei, 2021-01-08:
@mikemountain thanks Mike!  I edited the files with your fixes.  It is now displaying WSH vs TB correctly now.  Thank you.
Would it be possible to rotate through the rest of the scheduled games too as if your favorite team is NOT in the playoffs?  Just a wish, but not sure if its an easy fix or not, again really appreciate the quick fix for this one! thanks!

PR #18 (2021-01-08): b/fix playoff teams
if your favourite team is in the playoffs, it will now take it into consideration - if not, the board will cycle through each playoff game

fixes #17