mikemountain/nfl-led-scoreboard #19: cant execute nhl setup

ceegmac, 2021-01-13:
Hi, just struggling getting my scoreboard setup after the updates.
  _______________  _____
  / __/ __/_  __/ / / / _ \
 _\ \/ _/  / / / /_/ / ___/
/___/___/ /_/  \____/_/

1.5.0
Traceback (most recent call last):
  File "nhl_setup.py", line 1605, in <module>
  File "nhl_setup.py", line 1518, in main
PermissionError: [Errno 13] Permission denied: 'config/.default/firstrun'
[1130] Failed to execute script nhl_setup
pi@raspberrypi:~/nhl-led-scoreboard $

basically that is what i am getting when i try to run setup. 
had it working before and i dont think my up date went well. and scoreboard boots it doesnt do anything. can get the test screen  "HELLO WORLD" to work though

any help is appreciated

thank you

mikemountain, 2021-01-13:
Think you have the wrong repo, this is the N**F**L scoreboard repo

ceegmac, 2021-01-13:
sorry about that.

But I am planning on making the NFL one before season. cant wait. thank you
for doing that

mikemountain, 2021-01-13:
No problem, just wanted to confirm. Enjoy the season, and go leafs go woo

closed 2021-01-13 by mikemountain

siebe41, 2021-01-13:
Fwiw make sure you mark the file as executable with chmod

ceegmac, 2021-01-13:
sorry to be daft but how do you do that

siebe41, 2021-01-13:
chmod +x yourfilename