mikemountain/nfl-led-scoreboard #20: Logos: Pixel Perfect?

necromancyr, 2021-01-16:
Great project!  With the season over, it'll prob. go dormant for a bit but wondering if you'd be interested in some more pixel-perfect/accurate logos for teams?  The blur on the Eagles logo jumped out at me and was thinking maybe doing some small scale (like the link below - from a BoardGameGeek microbadge) might be a good fit if tweaked.  

[image]

I believe this is what was done for the NHL board.   

From the code, it looks like you resize to 19x19 - is that for everything, only certain boards, etc?

mikemountain, 2021-08-12:
Hey! You were right about it going dormant. Some of the logos do look pretty rough, gotta agree. Pre-game they're resized to decent sizes to fill the space, but during the game they do all get resized to 19x19. 

I'd love cleaner ones (I did look at the BGG microbadges but scaled up they don't look fantastic on the boards either) but I've just been rolling with what we've got

mikemountain, 2021-09-02:
I added some 8-bit helmet logos in [this branch](https://github.com/mikemountain/nfl-led-scoreboard/tree/f/rotate-games). I'm still going to look into getting some actual 8-bit team logos made that will look the best but the helmets do look a bit nicer at times. more consistent too

mikemountain, 2021-09-15:
Logos are way better now in the new master branch, and there's an option for helmets.

closed 2021-09-15 by mikemountain