mikemountain/nfl-led-scoreboard #21: key error on superbowl teams?

machinasdei, 2021-01-28:
here's the output, just seemed to pop up recently: not sure how to fix this...

INFO (18:31:13): NFL Scoreboard - v0.0.1 (64x32)
INFO (18:31:15): Final State, waiting 6 hours
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 29, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 54, in __render_game
    self._draw_post_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 257, in _draw_post_game
    awaysize = self.screen_config.team_logos_pos[overview['awayteam']]['size']
KeyError: u'AFC'

breid1313, 2021-01-30:
Yup. This is the games dict that the NFL API is returning right now. New to this project as of today, so not sure if they are just slow to update their data with the team names this year, or if this is always the case with the super bowl.
```
{'awayid': u'31', 'awayscore': 0, 'awayteam': u'AFC', 'date': u'2021-01-31T05:00Z', 'down': None, 'homeid': u'32', 'homescore': 0, 'hometeam': u'NFC', 'name': u'AFC @ NFC', 'over': False, 'possession': None, 'quarter': 0, 'redzone': None, 'spot': None, ...}
```

breid1313, 2021-01-30:
@machinasdei if you want a silly (and not great looking) fix just to get your board back up and running again, you can run off my branch: https://github.com/breid1313/nfl-led-scoreboard/tree/superbowl-patch

All I did was add a logo for AFC and NFL and gave them the same config settings as ARI. Not ideal, but it'll run :)

mikemountain, 2021-01-30:
Oh shit, looks like this is for the All Star game, my bad! Didn't even know ESPN would put up scores for this, this didn't come up last year. 

@breid1313 would you mind making a PR for this and I'll merge it in / can I pull your work into this repo? Thanks!

breid1313, 2021-01-30:
Sure, I’ll be able to open one in about an hour or so. I would have opened it up last night but it’s a pretty quick and dirty fix so I wasn’t sure if there was a better alternative. 

You may want to take a look at the json config file, as I literally just copy pasted the settings for Arizona with no knowledge or regard to what the best values would be. Maybe I’ll put a picture along with the PR and let you decide.

mikemountain, 2021-01-30:
Yeah, no sweat bud - considering the All Star game this year is cancelled, not a massive deal. Just good to have it that the boards don't crash. Thanks!

breid1313, 2021-01-30:
Thanks the exact intent - just keep it running 👍

commit 820e343c (2021-01-30): Superbowl patch (#22)

* AFC and NFC logos, random configs

* alphabetize team configs

thanks @breid1313 ! closes #21
closed 2021-01-30 by mikemountain