mikemountain/nfl-led-scoreboard #23: No module named tzlocal/mfl_api_parser

KyleLosch10, 2021-02-07:
I followed the instructions and when I run the command to start the board, I get the following errors:

~/nfl-led-scoreboard $ sudo python main.py --led-gpio-mapping=adafruit-hat
Traceback (most recent call last):
  File "main.py", line 2, in <module>
    from data.scoreboard_config import ScoreboardConfig
  File "/home/pi/nfl-led-scoreboard/data/scoreboard_config.py", line 1, in <module>
    from utils import get_file
  File "/home/pi/nfl-led-scoreboard/utils.py", line 7, in <module>
    from tzlocal import get_localzone
ImportError: No module named tzlocal

Also. with other projects I have had to use python3 instead of python. when I try python3 I get this error:

~/nfl-led-scoreboard $ sudo python3 main.py --led-gpio-mapping=adafruit-hat
Traceback (most recent call last):
  File "main.py", line 6, in <module>
    from data.data import Data
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 3, in <module>
    import nfl_api_parser as nflparser
ModuleNotFoundError: No module named 'nfl_api_parser'

I also have the NHL board running on the same pi. I don't know if that has something to do with it

mikemountain, 2021-02-07:
Hm weird, you ran the install script? Can you try running `sudo pip install tzlocal` by itself instead of in the script and post pack if it succeeds, and if the board continues to fail?

KyleLosch10, 2021-02-07:
So. I have the NHL screen. I was pretty sure I had "sudo apt-get install git python-pip" step already accomplished and did not need to run it again

I was dead wrong

It works now!

Note to self kids. Read the full instructions and actually follow the full instructions before creating an issue. Thanks for your help!

closed 2021-02-07 by KyleLosch10