mikemountain/nfl-led-scoreboard #24: Off-season Blank?

WashCaps33, 2021-05-10:
Just now doing this project.  Is it normal that the LED stays blank during the off-season?

richmyers, 2021-05-13:
Form what I can tell but running the app......it looks like ESPN API is not responding.  Might be because we are not in season....Not sure.

pi@NFLScoreboard:~/nfl-led-scoreboard $ ./start_scoreboard.sh
INFO (20:03:49): NFL Scoreboard - v0.0.1 (64x32)
Error encountered getting game info, can't hit ESPN api
Error encountered getting game info, can't hit ESPN api
Error encountered getting game info, can't hit ESPN api
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 29, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 33, in __render_game
    gametime = datetime.strptime(self.data.game['date'], "%Y-%m-%dT%H:%MZ")
TypeError: 'NoneType' object has no attribute '__getitem__'

mikemountain, 2021-08-05:
This may have been related to #27 but I'll leave it open and remind myself to make a generic off-season screen