mikemountain/nfl-led-scoreboard #29: Board doesn't transition from pregame->live or live->postgame without rerunning

tjhall9, 2021-08-06:
Running with `sudo ... &`, program will stay on the pregame screen until the script is restarted manually, same things happen postgame where the final score didn't appear until manual restart. Would love to leave it running and have it automatically transition between states without user input.

WashCaps33, 2021-08-06:
Either I’m doing something wrong or something is off, but I know the WFT have a game tonight and I am still getting errors.  So I’m lost.

tjhall9, 2021-08-06:
The next preseason games aren't until 8/12

mikemountain, 2021-08-06:
@tjhall9 I noticed this too, thanks for confirming it's the same with the final score/endgame situation. Will definitely look into it for next week's games.

@WashCaps33 have you pulled the latest master branch? I see the end screen for last night's game when I boot up.

WashCaps33, 2021-08-06:
Well damn.  Shows how much I care about the preseason. Lol sorry about that

WashCaps33, 2021-08-12:
Trying this today seeing as it is a game day for the WFT and while I get no error messages, I do get a “Schedule State, waiting 30” message and nothing happens.  Is this normal?

tjhall9, 2021-08-12:
Yeah that's normal. It may not actually show the live game without an update once the game starts though

WashCaps33, 2021-08-12:
Ahhh.  Thanks for the info!

mikemountain, 2021-08-12:
Think I've fixed this on this branch https://github.com/mikemountain/nfl-led-scoreboard/tree/b/fix-pre-post-game-transition-states but I'll be testing with the real game before merging it in, just to be sure

WashCaps33, 2021-08-13:
I must be doing something wrong.  Now that the game has started, it’s updating every 3 seconds, but nothing is appearing on the LED board.

mikemountain, 2021-08-13:
@WashCaps33 can you post the output of the script? You might need to kill and restart the script, my new bugfix isn't in yet and there was an issue with the game time in it

WashCaps33, 2021-08-13:
It just continuously updates.  The script works as it should, just nothing transmitting to the board.

pjockey, 2021-08-13:
What is the updating process? I tried the git commands, but got a bunch of errors.

tjhall9, 2021-08-14:
@mikemountain my board is still stuck without showing the final screen again, I wasn't home when the game started but it was running live when I got back and not sitting in scheduled. So it looks like it's halfway there

tjhall9, 2021-08-15:
@mikemountain transition to final state worked at the end of CAR-IND. Not sure why it didn't happen with the games previously

PR #30 (2021-08-21): fix pre post game transition states
Double checked today with the Bills-Bears game, it went from pregame countdown to live game to end screen properly. 

Closes #29 but I still need a way to kinda get to new games too. working on that
closed 2021-08-21 by mikemountain