mikemountain/nfl-led-scoreboard #32: Update Commands

xTrackstar04x, 2021-09-09:
Can you put what commands need to be run to update the NFL scoreboard in the readme?

mikemountain, 2021-09-09:
Hey, so if you're talking about the rotating game feature, it's not in the main branch yet as it's still in testing, but I can post here how to run off that test branch:

1. run `git stash` in the directory where you have the NFL scoreboard running
2. run `git fetch`
3. run `git checkout f/rotate-games`
4. run `cp config.json.example config.json`
5. run `nano config.json`
6. edit in your favourite team(s) in the preferred teams field
7. run the board as normal with `sudo python main.py` and whatever args you use after

mikemountain, 2021-09-09:
also, if things are breaking, periodically try running `git pull` to see if I've fixed them because things will almost certainly be breaking.

xTrackstar04x, 2021-09-10:
this is what I am getting when I run ```git stash``` and ```git fetch```

pi@raspberrypi:~/nfl-led-scoreboard $ git stash
Saved working directory and index state WIP on master: 820e343 Superbowl patch (#22)
pi@raspberrypi:~/nfl-led-scoreboard $ get fetch
-bash: get: command not found

mikemountain, 2021-09-10:
Running `sudo chown -R $(whoami) .git/` should fix that, try that and re-run `git fetch`

xTrackstar04x, 2021-09-10:
That worked! Thanks.

When I ran ```get fetch``` it came back with 3 new branches. Should I also run ```get checkout``` on the other 2 branches?

``` 
 820e343..8f324fa  master             -> origin/master
 * [new branch]      b/fix-pre-post-game-transition-states -> origin/b/fix-pre-post-game-transition-states
 * [new branch]      b/show-game-hotfix -> origin/b/show-game-hotfix
 * [new branch]      f/rotate-games     -> origin/f/rotate-games
pi@raspberrypi:~/nfl-led-scoreboard $ git checkout f/rotate-games
```

mikemountain, 2021-09-10:
Nope, running `git checkout` essentially switches all the files to a certain state. So the files in f/rotate-games won't match the files in master. Just as a heads up, there are still a decent amount of bugs in the f/rotate-games branch, so expect things to fail tomorrow - and if they do fail, wait about 10 mins and try running `git pull` to see if I've pushed an update

xTrackstar04x, 2021-09-10:
10-4. I'll let you know if I find anything as well

xTrackstar04x, 2021-09-12:
The board is not rotating for me. Any ideas?

mikemountain, 2021-09-12:
Can you post your config?

xTrackstar04x, 2021-09-12:
[image]

WashCaps33, 2021-09-12:
I keep getting this.  No idea what to do to resolve it.

WashCaps33, 2021-09-13:
Any reason I would get an error when trying to run git fetch?

mikemountain, 2021-09-13:
@WashCaps33 I can't help without any info on what error you're getting, but since this is an experimental branch that's still likely to run into bugs, hold off until I merge it into the main code (should be tomorrow) and then we'll go from there

WashCaps33, 2021-09-13:
Got it. Thanks.