mikemountain/nfl-led-scoreboard #33: Animations are broken with game rotation

mikemountain, 2021-09-10:
so the way it used to work was, with just one game always running, could easily track the changes in score. now with rotations, that's kinda been removed, so I gotta re-figure it out

pjockey, 2021-09-10:
Trying get this board up and running. The MLB and NHL boards both work fine. When I run `sudo python main.py` I get the following errors:

```
sudo python main.py
INFO (21:04:46): NFL Scoreboard - v0.0.1 (64x32)
INFO (21:04:47): Live State, checking every 5s
INFO (21:04:47): Refresh game overview
Traceback (most recent call last):
  File "main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 32, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 41, in __render_game
    self.__draw_game(self.data.current_game())
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 104, in __draw_game
    self._draw_live_game(game)
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 211, in _draw_live_game
    self.data.refresh_game()
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 36, in refresh_game
    self.game = self.choose_game()
AttributeError: Data instance has no attribute 'choose_game'
```
This also seems to lock the pi (3A+) making a hard boot necessary.

xTrackstar04x, 2021-09-10:
pretty sure you have to have flags on the command ```python main.py --led-gpio-mapping="adafruit-hat" --led-brightness=30 --led-cols=64```

pjockey, 2021-09-10:
> pretty sure you have to have flags on the command `python main.py --led-gpio-mapping="adafruit-hat" --led-brightness=30 --led-cols=64`

Yes that is what I have in Supervisor, yet it takes the Pi off of the network and locks it up tight. Wanted to see what was happening.

machinasdei, 2021-10-30:
> so the way it used to work was, with just one game always running, could easily track the changes in score. now with rotations, that's kinda been removed, so I gotta re-figure it out

any luck?