mikemountain/nfl-led-scoreboard #35: Why am I getting this when I try to sudo git pull

Bombsurace, 2021-09-18:
Updating 8f324fa..40d0d0e
error: Your local changes to the following files would be overwritten by merge:
        config.json
Please commit your changes or stash them before you merge.
Aborting

mikemountain, 2021-09-18:
Did you run the commands it told you to? You don't have to specifically set your email or name if you don't want to, git just wants to make sure changes are tracked properly.

Run the commands like it tells you to, can even be verbatim, and then try stashing again.

Bombsurace, 2021-09-18:
Got it, I thought I somehow was accessing something I wasn't suppose to.  Never had to add that stuff in there before so was super confused.  Followed the steps and all good.  You're awesome as always good sir, thanks for reaching out to dumb dumbs like me lol

closed 2021-09-19 by Bombsurace