mikemountain/nfl-led-scoreboard #38: Board Not Lighting Up on Install

CameronBarrett33, 2021-10-19:
I haven't been able to get the led board to light up on initial install, following the documentation provided. 

I first tried adding NFL to the same pi as my working MLB, then once I installed the NFL board, MLB stopped working also. I then tried on a fresh pi OS to isolate NFL, and still was not able to light up the board.

pjockey, 2021-10-21:
I think these scoreboards need to run in the Raspbian Minimal, no GUI or desktop. They need as much resources as possible to work properly.

If you need help with ssh part, lots of help around here.

CameronBarrett33, 2021-11-16:
I got it to work! I deleted the venv directory and that did the trick. I'm no expert but it seems that the venv directory had some python 3 in it possibly causing a version mismatch since MLB and NFL both use 2.7. @pjockey  thanks for the help!

pbullion, 2022-01-12:
I'm having this problem as well. @CameronBarrett33 after deleting the venv folder did you do anything else? ive tried a few things involving that and no luck