mikemountain/nfl-led-scoreboard #41: Preferred team not showing live, wrong team is displayed and does not rotate (GB Packers)

jediempi, 2021-11-28:
I am using the f/rotate games fork.

When the rotation is set to “true” and the only preferred is set to “false”, and the while preferred team live is set to “false” it does not display the correct team. My preferred team are Texas, and Green Bay, it is currently displaying the chargers and broncos and not GB. 

****Script used to launch the matrix:****
sudo python main.py --led-gpio-mapping="adafruit-hat-pwm" --led-brightness=40 --led-slowdown-gpio=2 --led-cols 64

**config.json**
{
        "preferred": {
                "teams": ["TX", "GB"]
        },
        "rotation": {
                "enabled": true,
                "only_preferred": false,
                "rates": {
                        "live": 15.0,
                        "final": 10.0,
                        "pregame": 10.0
                },
                "while_preferred_team_live": {
                        "enabled": false,
                        "during_halftime": true
                }
        },
    "scrolling_speed": 2,
    "use_helmet_logos": true,
    "debug": "true"
}

pjockey, 2021-11-29:
What happens is you add --rows=32 at the end of that script? I see that is missing after the cols 64. Also, there is a missing "=" sign in the `--led-cols 64`.

I'm not sure if the rows needs to be there or not. Just a shot.

jediempi, 2021-11-29:
Good eye on the = sign missing! I updated the code, but won't know if it worked until next Sunday when there are a bunch of games on. The matrix display shows the teams perfectly. Which leads me to believe the cols and rows are not the culprit, but I've seen crazier things.

I also added the --led-rows=32 bit to the script. I was reading through hzellers GitHub (https://github.com/hzeller/rpi-rgb-led-matrix) and it says the default for rows is 32. It can't hurt to hard code it in though. 

I'm using Supervisor to launch the NFL app along with the NBA, MLB, NHL, Tetris Clock, and Weatherstation. Is there a script like the NBA scoreboard that needs to run in the background (cron job) which gathers all of the teams and handles the rotation? Or is everything handled by launching main.py?

Thanks for the help and guidance!

pjockey, 2021-11-29:
I use Supervisor to switch between NFL, NHL and MLB. That's it at the moment. I believe my NFL config.json is the same as yours, except for the teams, and I don't have a rotations problem yet.

Blueice119, 2022-10-25:
> Good eye on the = sign missing! I updated the code, but won't know if it worked until next Sunday when there are a bunch of games on. The matrix display shows the teams perfectly. Which leads me to [...]

Hey were you able to figure out your issue? I'm having the same problem.

jediempi, 2022-10-25:
Unfortunately no. I still can't get the live games to cycle and stop when the GB packers are live. I have to choose to either have it cycle all games always, or set it to only my preferred team and turn live cycling off.

I went so far as to delete and reinstall the nfl repo and still have the issue.

mikemountain, 2022-10-25:
Hey yeah sorry guys, this is still a bug and I've been pretty overwhelmed
IRL to tackle it. Really want to fix it asap and have a plan on paper so I
will try to prioritise this asap

jediempi, 2022-10-25:
I'm super appreciative for all the work you've done! It's amazing to have the scoreboard going along side the game, and I'm a tech person so it's not too much of a hassle to edit the script when my team is live.

Would you be willing to share your thoughts on why this bug is happening?