mikemountain/nfl-led-scoreboard #44: 128 x 64 Screens

pjockey, 2022-09-07:
Just wondering if there is any problem running the script on the 128 x 64 screen. I tried to run it with this

`sudo python main.py --led-gpio-mapping=adafruit-hat-pwm --led-cols=128 --led-rows=64 --led-panel-type=FM6126B --led-rgb-sequence=BGR --led-row-addr-type=3`

mikemountain, 2022-09-07:
So so far, it will only work with 64 x 32 - it shouldn't be a big undertaking to tweak it for 128 x 64 though, so I'll try to get out something quickly

Irishking03, 2023-09-11:
Any luck with getting this to work.. I need to make this board bigger. I am currently running a 192x96 board for mlb and would love to display a 128x32 at minimum.

tarrant33, 2023-09-26:
Would gladly donate for this added here + fantasy code as well

RyanSwan1981, 2025-08-30:
Has anyone made any progress on this ask? I'm using four screens linked with the following flags:

--led-chain=1  
--led-parallel=2 
--led-rows=32 
--led-cols=128

This setup works perfectly with the MLB project. But I can't for the life of me make this work for the NFL project. I've been digging around the various .py files trying to see if I can just force this into a mode that utilizes all four screens. But no luck so far.