mikemountain/nfl-led-scoreboard #46: wont display anything

Slythy, 2022-09-15:
I cant seem to get the scoreboard working. I originally had issues with the ledmatrix repository and was able to fix that by downloading the mlb scoreboard to get the right python3 files i needed. Then I ran into the no API data which someone posted a fix for recently. 

I now have an error at the end that just says OSError: cannot open resource. The screen blips with 2 LED's when I run the program but then nothing. Any ideas?

Slythy, 2022-09-19:
@mikemountain any ideas?

mikemountain, 2022-09-19:
hey sorry! have been out of town recently and unable to do any testing.

Can you explain the "I originally had issues with the ledmatrix repository" part? What happens when you run `sudo ./install.sh` ?

Slythy, 2022-09-19:
@mikemountain no worries! just trying to get something running before football season really takes off.

 It gave me python errors /missing RGB LED issues. Those disappeared when I installed the MLB version of the scoreboard so something on the backend wasnt happy...    Everything made sense when I did it and looked like it was installing and running.

Currently it now just says "OSError: cannot open resource" like the picture shows. I dont know why. 2 LED's on the board just do a quick flash and then nothing happens. 

I was thinking about reinstalling the OS and trying again.

mikemountain, 2022-09-19:
Interesting. Can you post the outputs of 
`ping 8.8.8.8` 
`sudo apt-get update`
`sudo apt-get upgrade`
?

Slythy, 2022-09-19:
pings normal

the first upgrade and update had a TON in it... I reran it and now it doesnt have any updates to do... I tried relaunching the scoreboard still get an error. I dont know how to get past the OSError. I dont have any issues with the MLB scoreboard. 

The python-pip says its no longer available but i have the python3-pip.

donny@raspberrypi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
donny@raspberrypi:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following packages have been kept back:
  raspberrypi-ui-mods userconf-pi
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
donny@raspberrypi:~ $

donny@raspberrypi:~ $ sudo python nfl-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat-pwm --led-rows=32 --led-cols=64
/home/donny/nfl-led-scoreboard/renderer/main.py:365: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while x is not 3:
/home/donny/nfl-led-scoreboard/renderer/main.py:377: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while x is not 3:
/home/donny/nfl-led-scoreboard/renderer/main.py:398: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while x is not 3:
Suggestion: to slightly improve display update, add
isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
INFO (10:28:33): NFL Scoreboard - v1.0.0 (64x32)
Traceback (most recent call last):
  File "/home/donny/nfl-led-scoreboard/main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/donny/nfl-led-scoreboard/renderer/main.py", line 25, in __init__
    self.font = ImageFont.truetype("fonts/score_large.otf", 16)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 852, in truetype
    return freetype(font)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 849, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 209, in __init__
    self.font = core.getfont(
OSError: cannot open resource
donny@raspberrypi:~ $

mikemountain, 2022-09-19:
Just to make sure, did you do the pwm modification to the board?
Can you try `sudo ./install.sh` again now and post the output?

Slythy, 2022-09-20:
@mikemountain 

Yea for sure the mod is done, its working with the MLB scoreboard which is on a different sd card

So it doesnt do anything unless i do the chmod  before

donny@raspberrypi:~ $ cd nfl-led-scoreboard/
donny@raspberrypi:~/nfl-led-scoreboard $ sudo ./install.sh
sudo: ./install.sh: command not found
donny@raspberrypi:~/nfl-led-scoreboard $ sudo chmod +x install.sh
donny@raspberrypi:~/nfl-led-scoreboard $ sudo ./install.sh
Running rgbmatrix installation...
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Reading package lists... Done      
E: Release file for http://archive.raspberrypi.org/debian/dists/bullseye/InRelease is not valid yet (invalid for another 12h 24min 15s). Updates for this repository will not be applied.
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease is not valid yet (invalid for another 14h 54min 46s). Updates for this repository will not be applied.
make -C ./lib
make[1]: Entering directory '/home/donny/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/donny/nfl-led-scoreboard/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/donny/nfl-led-scoreboard/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/donny/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/donny/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/donny/nfl-led-scoreboard/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/donny/nfl-led-scoreboard/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/donny/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/donny/nfl-led-scoreboard/matrix/lib'
python setup.py build --build-lib .
running build
running build_py
running build_ext
building 'core' extension
arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I../../include -I/usr/include/python3.9 -c rgbmatrix/core.cpp -o build/temp.linux-armv7l-3.9/rgbmatrix/core.o -O3 -Wall
rgbmatrix/core.cpp: In function ‘PyObject* PyInit_core()’:
rgbmatrix/core.cpp:8633:38: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 8633 |   __pyx_type_9rgbmatrix_4core_Canvas.tp_print = 0;
      |                                      ^~~~~~~~
rgbmatrix/core.cpp:8644:43: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 8644 |   __pyx_type_9rgbmatrix_4core_FrameCanvas.tp_print = 0;
      |                                           ^~~~~~~~
rgbmatrix/core.cpp:8655:41: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 8655 |   __pyx_type_9rgbmatrix_4core_RGBMatrix.tp_print = 0;
      |                                         ^~~~~~~~
rgbmatrix/core.cpp:8661:48: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
 8661 |   __pyx_type_9rgbmatrix_4core_RGBMatrixOptions.tp_print = 0;
      |                                                ^~~~~~~~
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:97,
                 from rgbmatrix/core.cpp:4:
rgbmatrix/core.cpp: In function ‘int __Pyx_ParseOptionalKeywords(PyObject*, PyObject***, PyObject*, PyObject**, Py_ssize_t, const char*)’:
/usr/include/python3.9/cpython/unicodeobject.h:451:75: warning: ‘Py_ssize_t _PyUnicode_get_wstr_length(PyObject*)’ is deprecated [-Wdeprecated-declarations]
  451 | ine PyUnicode_WSTR_LENGTH(op) _PyUnicode_get_wstr_length((PyObject*)op)
      |                                                                       ^

/usr/include/python3.9/cpython/unicodeobject.h:261:7: note: in expansion of macro ‘PyUnicode_WSTR_LENGTH’
  261 |       PyUnicode_WSTR_LENGTH(op) :                    \
      |       ^~~~~~~~~~~~~~~~~~~~~
rgbmatrix/core.cpp:9039:22: note: in expansion of macro ‘PyUnicode_GET_SIZE’
 9039 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                      ^~~~~~~~~~~~~~~~~~
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/python3.9/cpython/unicodeobject.h:262:52: warning: ‘Py_UNICODE* PyUnicode_AsUnicode(PyObject*)’ is deprecated [-Wdeprecated-declarations]
  262 |       ((void)PyUnicode_AsUnicode(_PyObject_CAST(op)),\
      |                                                    ^
[... 327 lines cut ...]
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
libxml2-dev is already the newest version (2.9.10+dfsg-6.7+deb11u2).
libxslt1-dev is already the newest version (1.1.34-4+deb11u1).
[... cut ...]
Requirement already satisfied: pytz-deprecation-shim in /usr/local/lib/python3.9/dist-packages (from tzlocal) (0.1.0.post0)
Requirement already satisfied: tzdata in /usr/local/lib/python3.9/dist-packages (from pytz-deprecation-shim->tzlocal) (2022.2)
make: *** No targets specified and no makefile found.  Stop.
If you didn't see any errors above, everything should be installed!
Installation complete! Play around with the examples in matrix/bindings/python/samples to make sure your matrix is working.
donny@raspberrypi:~/nfl-led-scoreboard $

mikemountain, 2022-09-20:
Hmmm that's really weird. I'm wondering if it is an OS or even SD card thing, I can't say I've seen this before and don't know what could be causing it

Slythy, 2022-09-20:
I am reinstalling everything and going from there. so the current build shouldnt have any issues? do i need to install the python3-pip since the python-pip doesnt exist anymore?

mikemountain, 2022-09-27:
Hi @Slythy , I've updated the readme - if you're reinstalling everything from scratch, can you follow the instructions exactly in this section https://github.com/mikemountain/nfl-led-scoreboard#software-installation and once a step fails, stop and update this issue with the output? `python-pip` does still exist to my knowledge so I'd like to trace your exact steps