mikemountain/nfl-led-scoreboard #47: Animations not showing

maycl21, 2022-09-20:
After a bit of back and forth, i've gotten to a point where everything seems to work as it should and i'am very happy with the result.

The only issue i'm having, is that none of the animations are showing. Unfortunatly i was not able to see the terminal during the last games so im not sure if its giving me any helpful error messages when trying to load an animation. is there any known issues or common mistakes, maybe just a missing flag or something like this? 

Thanks.

maycl21, 2022-09-20:
Also the refresh rate seems to drop while the game is running to about once every 30-40 sec. Wich by itself isn't much of a problem for me but i am wondering if this might be the cause of the missing animations?

mikemountain, 2022-09-27:
hey, so I disabled the animations when I implemented the game rotation feature. I'm intending on bringing them back, but I'm still fighting with getting it working without interrupting the rotations