mikemountain/nfl-led-scoreboard #51: 64x64 board

Irishking03, 2023-07-30:
Just wondering if there was a 64x64 board option.

alexanderthebadatcoding, 2023-10-13:
Did you try adding `--led-rows=64 --led-cols=64`

Irishking03, 2023-10-13:
I belive I did. I thought maybe somewhere in the configed file it could be
adjusted.

alexanderthebadatcoding, 2023-10-13:
I edited the utils.py so I don't have to add the stuff at the end and can just run the main.py file.

`# Options for the rpi-rgb-led-matrix library
  parser.add_argument("--led-rows", action="store", help="Display rows. 16 for 16x32, 32 for 32x32. (Default: 32)", default=~32~64, type=int)
  parser.add_argument("--led-cols", action="store", help="Panel columns. Typically 32 or 64. (Default: 64)", default=64, type=int)`

Irishking03, 2023-10-13:
So just add the led cols 64 and led rows 64

RyanSwan1981, 2025-08-30:
Did your adjustment work for you? What was your results?