mikemountain/nfl-led-scoreboard #52: New Wifi, how to change?

Bombsurace, 2023-08-22:
Wasn't even thinking about it when it happened, but had to get new Router, so have a new name and password.  I currently cannot connect to the wifi portion due to the name and password change.

Am I screwed, or can I adjust it in the code on the micro sd card?  Is there another way to connect?  I guess I am confused, unless I can find/change in the hardcode, I am guessing I may have to reset the card and do everything from scratch once again?!

pjockey, 2023-08-22:
Can you get to the Terminal? If so, try `sudo raspi-config`. Should be able to get to the SSID and pasword.

Bombsurace, 2023-08-22:
I guess that's the issue, I was always connecting to the terminal with the wifi, but the wifi changed.  So it just wont connect.  I guess i should look up how to hardwire into terminal if possible

mikemountain, 2023-08-22:
I believe the pi's have an HDMI port and a USB port; you might need to open
up your case to use these but this way you'd have direct access to the
terminal

Bombsurace, 2023-08-22:
Aww, wasnt even thinking about that.  Thanks for the heads up!  Just have to find my small hdmi cord now.  Appreciate the heads up and replies!

closed 2023-08-22 by Bombsurace