mikemountain/nfl-led-scoreboard #56: Game rotation doesn't seem to work

btkircher, 2023-11-12:
Can you clarify what's required to get the games to rotate? I can only get the display to show one game at a time.

alexanderthebadatcoding, 2023-11-13:
Here is what I did to get it to work:

`	"rotation": {
		"enabled": true,
		"only_preferred": false,
		"rates": {
			"live": 15.0,
			"final": 10.0,
			"pregame": 10.0
		},
		"while_preferred_team_live": {
			"enabled": true,
			"during_halftime": true
		}
	},`

You have to change the game rotation to enabled: true, and the while preferred team Live settings in the config.json as well. Otherwise it just stayed on one game for me.

btkircher, 2023-11-13:
Oh, that's awesome, @alexanderthebadatcoding . That was the fix to get things rotating.