mikemountain/nfl-led-scoreboard #59: nfl_api_parser ERROR

Irishking03, 2023-11-29:
I just installed the nfl board on a new raspberry pi and this is the error I get using this argument in terminal

sudo python main.py --led-gpio-mapping=regular --led-cols=64 --led-rows=32 --led-brightness=60 --led-slowdown-gpio=2

Here is the ERROR:

File "/home/pi/nfl-led-scoreboard/main.py", line 6, in <module>
    from data.data import Data
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 3, in <module>
    import nfl_api_parser as nflparser
ModuleNotFoundError: No module named 'nfl_api_parser'

alexanderthebadatcoding, 2023-11-29:
Try sudo **python3** main.py. ....

Irishking03, 2023-11-30:
nope that doesn't work either @alexanderthebadatcoding 

pi@raspberrypi:~/nfl-led-scoreboard $ sudo python3 main.py --led-gpio-mapping=regular --led-brightness=60 --led-slowdown-gpio=4 --led-rows=32 --led-cols=64

Traceback (most recent call last):
  File "/home/pi/nfl-led-scoreboard/main.py", line 6, in <module>
    from data.data import Data
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 3, in <module>
    import nfl_api_parser as nflparser
ModuleNotFoundError: No module named 'nfl_api_parser'

pjockey, 2023-11-30:
Did your install install the nfl_api_parser?  Try `pip3 install nfl_api_parser`.

Irishking03, 2023-11-30:
@pjockey yes it's there.

alexanderthebadatcoding, 2023-11-30:
I use python3 for the script but in the "/home/pi/nfl-led-scoreboard/data/data.py" file I added:

`import` data.nfl_api_parser as nflparser

Irishking03, 2023-11-30:
@alexanderthebadatcoding 

/home/pi/nfl-led-scoreboard/data/data.py

from datetime import datetime, timedelta
import time as t
import nfl_api_parser as nflparser
import debug

NETWORK_RETRY_SLEEP_TIME = 10.0

class Data:
    def __init__(self, config):
        # Save the parsed config
        self.config = config

        # Flag to determine when to refresh data
        self.needs_refresh = True

        self.helmet_logos = self.config.helmet_logos
        
        # Parse today's date and see if we should use today or yesterday
        self.get_current_date()
        # Fetch the teams info
        self.refresh_games()

        # self.playoffs = nflparser.is_playoffs()
        # self.games = nflparser.get_all_games()
        # self.game = self.choose_game()
        # self.gametime = self.get_gametime()

        # What game do we want to start on?
        self.current_game_index = 0
        self.current_division_index = 0
        # self.scores = {}

    def get_current_date(self):
        return datetime.utcnow()
    
    def refresh_game(self):
        self.game = self.choose_game()
        self.needs_refresh = False

    def refresh_games(self):
        attempts_remaining = 5
        while attempts_remaining > 0:
            try:
                all_games = nflparser.get_all_games()
                if self.config.rotation_only_preferred:
                    self.games = self.__filter_list_of_games(all_games, self.config.preferred_teams)
                # if rotation is disabled, only look at the first team in the list of preferred teams
                elif not self.config.rotation_enabled:
                    self.games = self.__filter_list_of_games(all_games, [self.config.preferred_teams[0]])
                else:
                    self.games = all_games

                self.games_refresh_time = t.time()
                self.network_issues = False
                break
            except Exception as e:
                self.network_issues = True
                debug.error("Networking error while refreshing the master list of games. {} retries remaining.".format(attempts_remaining))
                debug.error("Exception: {}".format(e))
                attempts_remaining -= 1
                t.sleep(NETWORK_RETRY_SLEEP_TIME)
            except ValueError:
                self.network_issues = True
                debug.error("Value Error while refreshing master list of games. {} retries remaining.".format(attempts_remaining))
                debug.error("ValueError: Failed to refresh list of games")
                attempts_remaining -= 1
[... 59 lines cut ...]
    def __filter_list_of_games(self, games, teams):
        return list(game for game in games if set([game['awayteam'], game['hometeam']]).intersection(set(teams)))

    # def __game_index_for(self, team_name):
    #     team_index = 0
    #     print(self.games)
    #     # team_idxs = [i for i, game in enumerate(self.games) if team_name in [game.awayteam, game.hometeam]]
    #     for game in enumerate(self.games):
    #         print(game)
    #     return team_index

    def __next_game_index(self):
        counter = self.current_game_index + 1
        if counter >= len(self.games):
            counter = 0
        return counter

    #
    # Debug info

    # def print_overview_debug(self):
    #     debug.log("Overview Refreshed: {}".format(self.overview.id))
    #     debug.log("Pre: {}".format(Pregame(self.overview, self.config.time_format)))
    #     debug.log("Live: {}".format(Scoreboard(self.overview)))
    #     debug.log("Final: {}".format(Final(self.current_game())))

Irishking03, 2023-11-30:
@alexanderthebadatcoding I have that already

from datetime import datetime, timedelta
import time as t
import nfl_api_parser as nflparser
import debug

NETWORK_RETRY_SLEEP_TIME = 10.0

class Data:
    def __init__(self, config):
        # Save the parsed config
        self.config = config

        # Flag to determine when to refresh data
        self.needs_refresh = True

        self.helmet_logos = self.config.helmet_logos
        
        # Parse today's date and see if we should use today or yesterday
        self.get_current_date()
        # Fetch the teams info
        self.refresh_games()

        # self.playoffs = nflparser.is_playoffs()
        # self.games = nflparser.get_all_games()
        # self.game = self.choose_game()
        # self.gametime = self.get_gametime()

        # What game do we want to start on?
        self.current_game_index = 0
        self.current_division_index = 0
        # self.scores = {}

Irishking03, 2023-11-30:
@alexanderthebadatcoding I have that already in the data.py 

from datetime import datetime, timedelta
import time as t
import nfl_api_parser as nflparser
import debug

NETWORK_RETRY_SLEEP_TIME = 10.0

Irishking03, 2023-11-30:
@alexanderthebadatcoding NEW ERROR NOW..

 sudo python main.py --led-gpio-mapping=regular --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64Suggestion: to slightly improve display update, add
	isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
INFO (15:07:19): NFL Scoreboard - v1.0.0 (64x32)
something bad? Expecting value: line 1 column 1 (char 0)
something bad? Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/home/pi/nfl-led-scoreboard/main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 32, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 41, in __render_game
    self.__draw_game(self.data.current_game())
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 79, in current_game
    return self.games[self.current_game_index]
TypeError: 'NoneType' object is not subscriptable

Irishking03, 2023-11-30:
NEVERMIND.... I GOT IT
 Now to play with the renderer files

THANK YOU SO MUCH

Irishking03, 2023-11-30:
@mikemountain nah man your good... Im playing with renders now and adjusting to fit 196x92 board
[image]

Irishking03, 2023-11-30:
[image]

pjockey, 2023-11-30:
9 panels, 64x32? Nice.

Irishking03, 2023-11-30:
[image]
[image]
[image]

Irishking03, 2023-11-30:
@alexanderthebadatcoding you have a picture of the MLS one. Love to see that.. ALSO this is the new error I get. The board shuts down after a little while and now when trying to redo the code I get

~/nfl-led-scoreboard $ sudo python main.py --led-gpio-mapping=regular --led-brightness=60 --led-slowdown-gpio=2 --led-chain=3 --led-parallel=3

Suggestion: to slightly improve display update, add
	isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
INFO (18:09:31): NFL Scoreboard - v1.0.0 (192x96)
Traceback (most recent call last):
  File "/home/pi/nfl-led-scoreboard/main.py", line 30, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 32, in render
    self.__render_game()
  File "/home/pi/nfl-led-scoreboard/renderer/main.py", line 41, in __render_game
    self.__draw_game(self.data.current_game())
  File "/home/pi/nfl-led-scoreboard/data/data.py", line 79, in current_game
    return self.games[self.current_game_index]
IndexError: list index out of range

Irishking03, 2023-12-01:
I started over again. I think I was doing to much at once. But got it back working

Irishking03, 2023-12-01:
[image]

alexanderthebadatcoding, 2023-12-01:
You should check the troubleshooting section, https://github.com/hzeller/rpi-rgb-led-matrix/#troubleshooting and CPU Usage: https://github.com/hzeller/rpi-rgb-led-matrix/#cpu-use

Irishking03, 2023-12-02:
@alexanderthebadatcoding im good now I am back up and running. I had some of the false and true statements mixed up I think. Also it did not help that I was doing PHI instead of PHL or whatever it was.

alexanderthebadatcoding, 2023-12-07:
@Irishking03 if the NHL api is still down you can try this:
https://github.com/alexanderthebadatcoding/nhl-led/

I think I got it working.

pjockey, 2023-12-07:
> https://github.com/alexanderthebadatcoding/nhl-led/

Is this a python2 project? During the install, the installer could not find python-pillow. Using Bullseye OS. Lots of errors.

alexanderthebadatcoding, 2023-12-08:
> > https://github.com/alexanderthebadatcoding/nhl-led/
> 
> 
> 
> Is this a python2 project? During the install, the installer could not find python-pillow. Using Bullseye OS. Lots of errors.

I don't think so. But it is kind of a mess of different projects combined. So that could be the issue. I need to change the readme. Are you able to run it without installing? Just ’sudo python3 main.py’

I am just using the default raspberry pi lite OS.

pjockey, 2023-12-08:
the scrpit runs with `sudo python main.py` prior to attempting the ./install.sh. only python3 installed. The script runs, but the screen is giberous. Just a bunch of bluey green leds light up.

alexanderthebadatcoding, 2023-12-09:
I think I changed the settings in the util.py file, I added the panel type to: default="FM6126A" you should be able to delete that, and see if that helps. also change the gpio-mapping settings, if you're not using the adafruit hat.

(I'll update the code too)