mikemountain/nfl-led-scoreboard #60: Getting to rotate

Irishking03, 2023-12-01:
@mikemountain or anyone

Where is the setting to rotate the game. I currently have this in the config.json

{
	"preferred": {
		"teams": ["PHI", "DAL", "WASH", "NYG"]
	},
	"rotation": {
		"enabled": true,
		"only_preferred": false,
		"rates": {
			"live": 15.0,
			"final": 10.0,
			"pregame": 10.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_halftime": false
		}
	},
    "scrolling_speed": 2,
    "use_helmet_logos": false,
    "debug": "true"
}

BUT I CANT GET THE GAMES TO ROTATE LIKE THEY NORMALLY DO.

alexanderthebadatcoding, 2023-12-01:
try changing 
"while_preferred_team_live": {
"enabled": **true**,
"during_halftime": **true**

Irishking03, 2023-12-01:
@alexanderthebadatcoding WOULD HELP IF I Wasn't and idiot. I fixed that and also the PHL for eagles is PHI (SMACKS HEAD).. However now the eagles upcoming game comes up but I also have commanders, giants and dallas as preferred. But not rotating.

Irishking03, 2023-12-01:
It Looks like whatever I put in Preferred team as the last team (example  "team": ["PHI", "LAR", "DAL", "NYG"]) will populate only. Like in this example is NYG.. If i remove NYG just Dallas populates.. So wierd.

Irishking03, 2023-12-01:
I don't know what I did but it is rotating now. 

{
	"preferred": {
		"teams": ["PHI", "WAS"]
	},
	"rotation": {
		"enabled": true,
		"only_preferred": false,
		"rates": {
			"live": 15.0,
			"final": 10.0,
			"pregame": 10.0
		},
		"while_preferred_team_live": {
			"enabled": true,
			"during_halftime": false
		}
	},
    "scrolling_speed": 2,
    "use_helmet_logos": false,
    "debug": "true"
}

alexanderthebadatcoding, 2023-12-01:
I think there is a bug in the code, there's a post saying that the while preferred_team_live settings can cause an Issue. I think what is happening is that the code checks if a proffered team is playing, but then shows the first result returned from the API: http://site.api.espn.com/apis/site/v2/sports/football/nfl/scoreboard, which for this week is Dallas vs Seattle. 

thats what my guess is. I just made sure to have rotation "enabled": true, and during_halftime: true. They it will just always rotate the games shown.

Irishking03, 2023-12-09:
Now issue is the board with randomly turn off after a period of time. I have to log back in the pi (SSH) and re-enter the code.

alexanderthebadatcoding, 2023-12-09:
You could try to set a Supervisor task to run try script continuously, and if it crashes then restart it. 
`sudo apt-get install -y supervisor`
`sudo service supervisor start`
More information here: https://serversforhackers.com/c/monitoring-processes-with-supervisord

Also I created a script that switches between the NFL and NBA scoreboard etc.
But you might need to edit the settings a little bit to get it working.
What I did was created and led.py script and then set that to run on boot.

and then the led.py script has:

```
import subprocess
import os

# Define the paths to the two scripts you want to switch between
nfl = '/home/pi/nflscore/main.py'
nba = '/home/pi/nba-led-scoreboard/main.py'

def run_script(script_path, duration):
    try:
        # Run script
        process = subprocess.Popen(["python3", script_path])
        # print("Loading " + script_path)

        # Wait for the specified duration
        time.sleep(duration)
        # Terminate the subprocess
        process.terminate()

    except Exception as e:
        print(f"An unexpected error occurred: {e}")
        # Handle the unexpected error, for example, exit the script
        exit(1)

while True:
        # Run NBA
        print("Showing NBA Scoreboard")
        run_script(nba, 200)

         # Run NFL
         print("Showing NFL Scores")
         run_script(nfl, 100)
```

However for this to work I did have to edit the renderer files to change the paths for the logos. But maybe there is a way around that.