mikemountain/nfl-led-scoreboard #62: Install Error

bbgun75, 2024-07-13:
Getting this error when trying to install. I was able to get the MLB version running fine.

fatal: destination path 'matrix' already exists and is not an empty directory.
Running rgbmatrix installation...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease                  
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease                 
Reading package lists... Done                              
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python2.7-dev
E: Couldn't find any package by glob 'python2.7-dev'
E: Couldn't find any package by regex 'python2.7-dev'
E: Unable to locate package python-pillow
make -C ./lib
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
python setup.py build --build-lib .
rgbmatrix/core.cpp:162:12: fatal error: longintrepr.h: No such file or directory
  162 |   #include "longintrepr.h"
      |            ^~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
make[1]: *** [Makefile:67: build-python] Error 1
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/bindings/python'
make: *** [Makefile:28: build-python] Error 2
make -C ./lib
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/home/brian/nfl-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/lib'
python setup.py build --build-lib .
rgbmatrix/core.cpp:162:12: fatal error: longintrepr.h: No such file or directory
  162 |   #include "longintrepr.h"
      |            ^~~~~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
make[1]: *** [Makefile:67: build-python] Error 1
make[1]: Leaving directory '/home/brian/nfl-led-scoreboard/matrix/bindings/python'
make: *** [Makefile:28: build-python] Error 2
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Installing required dependencies. This may take some time (10-20 minutes-ish)...
HEAD is now at 0aa9f25 Update install.sh
Already on 'master'
Your branch is up to date with 'origin/master'.
Already up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
libxml2-dev is already the newest version (2.9.14+dfsg-1.3~deb12u1).
libxslt1-dev is already the newest version (1.1.35-1).
The following packages were automatically installed and are no longer required:
  libcamera0.2 libraspberrypi0 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
make: *** No targets specified and no makefile found.  Stop.
If you didn't see any errors above, everything should be installed!
Installation complete! Play around with the examples in matrix/bindings/python/samples to make sure your matrix is working.

pjockey, 2024-07-14:
Which version of python are you using?

bbgun75, 2024-07-14:
3

pjockey, 2024-07-14:
Python3.11 or 3.12?
The newest version of Python requires all of the scripts be run inside of the virtual environment. Not sure how far along this program is in regards to this.

bbgun75, 2024-07-14:
3.11.2

pjockey, 2024-07-14:
The version of python my be your problem. I see nothing has been update for years in Github for the NFL program. I'l bet it oes not run on 3.11. 

I use to have it running, but did not even try for the 2023 season. Send a DM to the creator to see if there is any progress.

alexanderthebadatcoding, 2024-07-19:
I had a similar issue when Installing. I believe I just deleted the Matrix folder and then ran the installation again. My RPI is running on the older OS and I have python 3.7

bbgun75, 2024-07-19:
Still not working. I'm running a Raspberry Pi 4 B if that makes a
difference.