mikemountain/nfl-led-scoreboard #66: Multiple Linked Screens / Larger Resolution

RyanSwan1981, 2025-08-30:
Posting this here for visibility:

Has anyone made any progress on the ability to use other resolutions other than the default 64 x 32? 

I'm using four 64 x 32 screens linked with the following flags:

--led-gpio-mapping=regular (adafruit triple led matrix bonnet)
--led-chain=1
--led-parallel=2
--led-rows=32
--led-cols=128

This setup works perfectly with the MLB project. But I can't for the life of me make this work for the NFL project. I've been digging around the various .py files trying to see if I can just force this into a mode that utilizes all four screens to act as one large screen. But no luck so far.

RyanSwan1981, 2025-08-31:
Okay....for anyone still interested in this project, I think I have cracked the long awaited resolution nut. Will post the new main.py script once I have tested it on a live game day.

RyanSwan1981, 2025-08-31:
Next order of business is to fix the logos so that the black pixels around the helmets are actually black/off

alexanderthebadatcoding, 2025-09-01:
> Next order of business is to fix the logos so that the black pixels around the helmets are actually black/off

This might be solved by adjusting the overall brightness. Also you can write a python script that checks the espn NFL API and downloads all the modern logos and adds a black background and sets them to 500x500px.

RyanSwan1981, 2025-09-02:
> This might be solved by adjusting the overall brightness. Also you can write a python script that checks the espn NFL API and downloads all the modern logos and adds a black background and sets them [...]

So I ended up taking the brute force approach and fixed the color values in Photoshop, including some of the helmets that weren't displaying properly due to their own color values. I can upload those files to my own git page.

alexanderthebadatcoding, 2025-09-02:
Feel free to look here if you need any other logos. [Sports Logos Repo](https://github.com/alexanderthebadatcoding/Sports-Logos)You can pretty easily display other sports by changing the ESPN api. So I experimented with a few other sports.

closed 2025-09-07 by RyanS-1981